BerkeleyLab/Marble #1: Bank voltages incompatible with Kintex

ldoolitt, 2020-01-10:
Kintex banks 32 and 33 are HP banks, limited to 1.8V.
This design has them connected to 2.5V supplies.

I discovered this when trying to get an overview of how the FPGA I/O is used:

bank 12:  HR  2.5V,  FMC2 LA 17-33
bank 13:  HR  2.5V,  FMC2 HB 00-20
bank 14:  HR  2.5V,  FMC1 LA 00-16  (plus config)
bank 15:  HR  2.5V,  FMC1 LA 17-33  (plus RGMII)
bank 16:  HR  3.3V,  local
bank 32:  HP  2.5V,  FMC2 HA 00-23
bank 33:  HP  2.5V,  FMC2 LA 00-16  (plus UART)
bank 34:  HP  1.35V, DDR3

gkasprow, 2020-01-10:
The same situation was with AFCK board. There is not much you can do - one of FMC has to be limited to 1.8V

ldoolitt, 2020-01-10:
Yes, I'm familiar with running FMC pins with a 1.8V FPGA bank.
If you only change the supply voltage on banks 32 and 33 down to 1.8V, you still have a likely unusable configuration, because FMC2 LA will be split between 1.8V and 2.5V.  And it's no good lowering the voltage on bank 12 to match, because HR banks need 2.5V to support LVDS, and most FMC use cases are heavy on the LVDS.

michalgaska, 2020-01-10:
The only solution is to move all HA signals to HP banks. Thus we will be able to set the power supply to 2.5V for all LA signals and 1.8V for HA signals.

gkasprow, 2020-01-10:
This is what I did for one of the banks in AFCK

michalgaska, 2020-01-16:
If you want to use HA and HB signals as an LVDS, the FPGA bank power supply doesn't matter (LVDS_25 or LVDS_18). 
It would be better to have a list of the FMC cards you want to use with Marble. I will check the incompatibilities and propose the solution.

gkasprow, 2020-01-16:
Usually, the LVDS is not an issue, but LVTTL.
 Some FMC boards use single-ended signalling which may not work with 1.8V.
In my FMC boards I always add level converter supplied from VADJ.

ldoolitt, 2020-03-04:
I scanned through a bunch of related FMC FPGA boards, and have a conclusion for how to proceed with the design.  Thankfully, it does not involve re-routing any data lines.

Looks to me like it's sufficient to connect Marble's banks 32 and 33 to 1.8V, and bank 13 to VIO_B_M2C.  This will work with Zest. It will also work with HPC mezzanine boards that allow 1.8V VADJ, as long as they don't try to use LVDS on bank B pins. Even that would be possible if the mezzanine managed to set VIO_B_M2C to 2.5V, allowed (and needed for LVDS) because in this scheme bank B is attached to HR pins.

Summary for review:
KC7K160T-2FFG676C, 5 HR banks and 3 HP banks

bank 12  HR  +2.5V      FMC2 LA 17-33
bank 13  HR  VIO_B_M2C  FMC2 HB 00-20
bank 14  HR  +2.5V      FMC1 LA 00-16  (plus config)
bank 15  HR  +2.5V      FMC1 LA 17-33  (plus RGMII)
bank 16  HR  +3.3V      Pmod, local
bank 32  HP  +1.8V      FMC2 HA 00-23
bank 33  HP  +1.8V      FMC2 LA 00-16  (plus UART)
bank 34  HP  +1.35V     DDR3

ldoolitt, 2020-09-10:
Based on longer email threads and on-line chats, we ended up allocating the three HP banks for DDR3 (in SO-DIMM form), and dropping FMC2 HB and native 3.3V signals.  All FMC signals now connect to 2.5V HR pins.  Summary:
KC7K160T-2FFG676C, 5 HR banks and 3 HP banks

bank 12  HR  +2.5V      FMC2 LA 00-16
bank 13  HR  +2.5V      FMC2 HA 00-23
bank 14  HR  +2.5V      FMC2 LA 17-33  (plus config, Pmod1)
bank 15  HR  +2.5V      FMC1 LA 00-16  (plus Pmod1)
bank 16  HR  +2.5V      FMC1 LA 17-33  (plus RGMII)
bank 32  HP  +1.5V      DDR3 
bank 33  HP  +1.5V      DDR3 (plus Pmod2)
bank 34  HP  +1.5V      DDR3

Where Pmod signals and other local features (not listed above) get routed through TXB0108 bidirectional level translators to get +3.3V compatibility.

closed 2020-09-10 by ldoolitt