BerkeleyLab/Marble #9: No capacitors around the ADN4600

michalgaska, 2020-09-24:
All capacitors on the clock signals are not soldered in. 
Capacitors have an attribute: -standalone but they must be soldered.

ldoolitt, 2020-10-02:
The capacitors in question are 24 x 100 nF (CC0201_100NF_6.3V_10%_X5R).  The ADN4600 is U2.
The caps are listed in the xy .pos file, but were mistakenly left out of the bom .csv file.
This omission was caused by invisible config attributes in the schematic (MGT_CLK_Crosspoint.sch).
The situation has not changed up to and including commit 623d3cc7, 2020-09-28.

Full list:
C30 C31 C32 C33 C34 C35 C82 C83 C84 C85 C86 C87 C88 C89 C94 C95 C96 C97 C98 C99 C100 C101 C288 C289

michalgaska, 2020-10-07:
Fixed in commit https://github.com/BerkeleyLab/Marble/commit/3606b1b726e1199c14f19437a218e99fa243ad47

closed 2021-10-06 by michalgaska