BerkeleyLab/Marble #22: ideal diode improvements

yetifrisstlama, 2021-01-20:
Simulating the ideal diode circuit around Q2 I stumbled across a bit of an issue.

When the input voltage is negative and the ideal diode should be blocking, there is a leakage path through Q1B, R107 and Q1A, shown with red arrow:

[image]

This is because emitter - base breakdown voltage of Q1A is only -5V and it sees the full input voltage there when the ideal diode should be blocking. 
Leakage current is only limited by R107, so about 17 mA at -24V input (5 mA @ -12V). 

... that's maybe a bit too much reverse current through the Q1A base to feel comfortable.

An easy fix would be to put a small (40 V, 0.1 A or so) diode as shown in green.

gkasprow, 2021-01-20:
Well, I added R107 to not break the Q1 in such a case.
The diode will break the circuit and Q2 won't switch on completely because the Q1 would need Q2 dropout to be higher than the diode dropout.  The easiest fix would be the Schottky diode placed between Q1B and GND so the Q1B base doesn't see any current during reverse connection.
It happened once that I connected 12V in the reversed way and did not observe any excessive current. Maybe at -24V, this circuit won't work.

yetifrisstlama, 2021-01-21:
>  The easiest fix would be the Schottky diode placed between Q1B and GND so the Q1B base doesn't see any current during reverse connection.

Could you make a quick sketch of this?

yetifrisstlama, 2021-01-21:
Ahh, thanks Greg, I see now how adding the green diode was a bad idea. It adds 0.2 V forward voltage drop to the ideal diode circuit, which is not what we want.

Here's another idea to keep the reverse current away from the Q1A base:

[image]

I've added `DX` (Schottky, 40 V, .2A) and `DY` (12 V Zener, only needed for output voltages > 12 V)

To reduce the magnitude of the reverse current I tried increasing R107. But this seems to introduce some unwanted hysteresis in the circuit. looks like there is some kind of matching required between R104, R107 and R108.

I came up with the values in the screenshot which seem to work okay for +- 30 V input voltage:
  * .5 mA leakage at -24 V input voltage
  * 30 mV forward voltage drop at 1 A
  * no reverse current into Q1A base