BerkeleyLab/Marble #23: QSFP1 lanes are swapped in two places

marmeladapk, 2021-02-05:
They are swapped both in FPGA_MGT schematic and in QSFP1 schematic. It would be less confusing to stick to one place for swapping lanes for readability.
[image]
[image]

ldoolitt, 2021-02-06:
That's true.  Someone else also noticed this recently.

It might be tricky to fix it in a way that guarantees no copper change, but I think I know the techniques needed.
If you want to beat me to it, and submit a pull request, that would be great!

ldoolitt, 2021-02-06:
@marmeladapk, please check my commit 3b2053a to branch issue_23, and see how it looks to you.

marmeladapk, 2021-02-11:
Looks good. AFAICT it doesn't change anything on PCB.