BerkeleyLab/Marble #37: Remove generated pdf files from repository

lerwys, 2022-02-04:
It's maybe interesting to remove generated files, such as pdfs ([Marble_Test_Guide.pdf](https://github.com/BerkeleyLab/Marble/blob/4e42f513a6ed428040387f1bcffafb63641b5238/docs/marble_test_guide/Marble_Test_Guide.pdf) and [Marble_User_guide.pdf](https://github.com/BerkeleyLab/Marble/blob/4e42f513a6ed428040387f1bcffafb63641b5238/docs/marble_user_guide/Marble_User_Guide.pdf)) , from checked-in files and put them only on each release, as it was done for [v1.1](https://github.com/BerkeleyLab/Marble/releases/tag/v1.1)

ldoolitt, 2025-04-07:
I appreciate the goal.  The trick is to make the generated files available and discoverable for the people who can use them. Now that we have a GitHub action set up (thanks, @mdavidsaver), can we leverage that, a bit like we did with Bedrock?

mdavidsaver, 2025-04-08:
In the context of github.com, the path of least friction for "rolling" generated documentation is probably [github pages](https://pages.github.com/).  eg. [an example](https://mdavidsaver.github.io/feed-core/) generated from [source](https://github.com/mdavidsaver/feed-core/tree/master/documentation).  For a few/large files, attaching to individual github releases may make more sense.

lerwys, 2025-04-08:
Please, see [add-sphinx-docs branch](https://github.com/BerkeleyLab/Marble/tree/add-sphinx-docs).

lerwys, 2025-04-11:
Larry suggested adding the Marble schematic to the generated docs. I updated the  [add-sphinx-docs branch](https://github.com/BerkeleyLab/Marble/tree/add-sphinx-docs) with that modification.

I've also followed @mdavidsaver idea to run DinD to ensure the same environment (i.e., Debian 12).

I'll open a MR soon.

ldoolitt, 2025-05-07:
Fixed by commit 81f31f9d.

closed 2025-05-07 by ldoolitt