BerkeleyLab/Marble #50: Marble v1.4

michalgaska, 2023-08-15:
TODO:

- [ ] Make sure that R147 is DNF
- [ ] R309 should be soldered on
- [ ] heatsink for U35 (PN: ATS-55170R-C1-R0)
- [ ] heatsink for U2 (clock mux)

ldoolitt, 2023-08-15:
I believe the design files say R147 is DNF, and R158 is installed, after commit 8614466.  Yes, please double-check.

ldoolitt, 2023-08-28:
I cross-checked that all these changes show up on the constructed marble-bom.csv file.  Yay!
Please add a manufacturer for the two new heat sinks:

* HS3 2100-20188-ND  ==  Watterott 20188 (Digi-Key 2100-20188-ND)
* HS2 ATS-55170R-C1-R0  ==  AdvancedThermalSolutions ATS-55170R-C1-R0 (Digi-Key ATS1298-ND)

michalgaska, 2023-08-28:
Please add a manufacturer for the two new heat sinks:
Done :)

ldoolitt, 2025-04-07:
fixed with commit eb2983a1 Aug 2023

closed 2025-04-07 by ldoolitt