BerkeleyLab/Marble #56: 1.4.3 design issue with AP62150

mdavidsaver, 2025-05-07:
Testing of a run of the 1.4.3 design finds that the new AP62150 +3.3P regulator part does not function correctly.  The power LED blinks once, then goes dark.  Subsequent testing finds that the +3.3P net has developed a short to ground.

As a candidate cause, the datasheet for the AP62150 lists the absolute max. for the Enable pin as -0.3 -> +6.0 .  Connecting Enable to +12 exceeds this.

mdavidsaver, 2025-05-28:
> As a candidate cause, the datasheet for the AP62150 lists the absolute max. for the Enable pin as -0.3 -> +6.0 . Connecting Enable to +12 exceeds this.

Cutting the trace between pins 1 and 5 of U18 allows this regulator to function correctly.

[image]

ldoolitt, 2025-05-29:
> [@michalgaska](https://github.com/michalgaska) [@ldoolitt](https://github.com/ldoolitt) the addition of L17 before the feedback network does not meet the goal of allowing this regulator to be tested with the load disconnected. Please consider moving R7 to the other side of L17.

Good catch!

michalgaska, 2025-05-29:
fixed

mdavidsaver, 2025-06-03:
After 1ec9219fe86c2e928d1bdf07c2a9727c188a74e1 looks ok.

[image]

mdavidsaver, 2025-06-03:
Looking at the layout I have one question.  Why does the (quite thin) trace to the top of R7 jump to the top layer?

[image]

mdavidsaver, 2025-06-03:
As opposed to say:

[image]

closed 2025-06-05 by ldoolitt