BerkeleyLab/Marble #60: Soldering issue with U48

mdavidsaver, 2025-06-03:
With a batch of 20x Marble 1.4.3, 11 boards failed the MGT loop back test with transceivers Y0 and/or Y1.  Also one failed on Y6.

mdavidsaver, 2025-06-03:
Inspection of the leadless U46 was difficult as the pads did not extend past the package body.

A resistance to ground test of pin 2 (power down) reported ~6.8k ohms on a functioning board, and ~90k ohms on a non-functional board.  Indicating a poor solder.

[image]

mdavidsaver, 2025-06-03:
@ldoolitt Noted that the footprint used for Marble <= 1.4.3 does not match the [OEM recommendation](https://www.diodes.com/part/view/PI3DBS16212/).

[image]

mdavidsaver, 2025-06-03:
> With a batch of 20x Marble 1.4.3, 11 boards failed the MGT loop back test with transceivers Y0 and/or Y1.

Removal and replacement remediated the Y0/Y1 MGT test failures.

> Also one failed on Y6.

This appears to be a separate issue.

mdavidsaver, 2025-06-03:
5435c09bb79a4c6682b5a34c2d88736f6f038417 enlarges all of the pads.

Below is screen capture where I have tried to overlay the old PI3DBS16212ZBBEX footprint above the new as a quick confirmation that the pads do not move.

[image]

ldoolitt, 2025-06-05:
Closing with the release of v1.4.4.
I hope this isn't unduly optimistic.

closed 2025-06-05 by ldoolitt